MartinL1/BMP280_DEV #5: How to change I2C pins during runtime

john-arvid, 2020-01-19:
Great library!

I wanted to change the I2C pins that is used for communication, the only place I found was in Device.cpp line 65, but is it possible to do during runtime?

MartinL1, 2020-01-31:
@john-arvid 

The BMP280_DEV library makes use of Arduino's Wire library. The Wire library uses Arduino boards' default SCL and SDA I2C pins which are fixed and depend on the board type, (Arduino Uno, Mega, Due, Zero, etc...).

Might I ask why you need to change the I2C pins during runtime, as the Wire library itself doesn't have this capability?

john-arvid, 2020-01-31:
@MartinL1 
Thanks for your reply.
Yes on Arduino boards the pins are fixed, but on ESP8266 board for example, the I2C pins are emulated in software and can be set to almost whatever pin.

Not sure if I worded myself correctly with "during runtime", but I want to set the pins in the code during setup for example. The reason for this is that some ESP8266 boards have limited pins broken out and some pins fit better depending on the sensor. (Non I2C sensors)

MartinL1, 2020-01-31:
@john-arvid 
Thanks for the clarification. It appears as though the ESP8266 is the exception to the rule. Are you able to use the default I2C pins on the ESP8266?: 

GPIO4 --> SDA
GPIO5 --> SCL

These pins should work with the BMP280_DEV library without any code modification.

john-arvid, 2020-01-31:
@MartinL1 
Yes I am able to use the default I2C pins (when I am using a board with those broken out), but how could I change those pins in the code? (Would like to do it in the code and not in the module)

MartinL1, 2020-02-01:
@john-arvid 
No problem, I'll add the ESP8266 functionality you require that will allow you to specify user defined SCL and SDA pins, together with some example code to the BMP280_DEV library in the next couple of days.

john-arvid, 2020-02-01:
@MartinL1 
Thank you! Greatly appreciated. 
I will leave the issue open so you can close it with the commit.

MartinL1, 2020-02-02:
@john-arvid 
I've made changes to allow user-defined pins with the ESP8266. In the Arduino sketch it's just necessary to specify the SDA and SCL pins numbers as arguments to the BMP280_DEV's constructor:

```
BMP280_DEV bmp280(6, 7);    // User-defined I2C pins, SDA: 6, SCL: 7
```

Other than that the sketch should remain the same. I've also included an example sketch that can be found in this library's _/example/_ directory.

Please could you tell me if this code works, as I don't have an ESP8266 board on which to test it? I'll make it available as an update once I've got confirmation that it's working OK.

john-arvid, 2020-02-02:
@MartinL1 
Thank you very much!
I will test in a couple of days when I get back home.

closed 2020-08-25 by MartinL1

kreamuhely, 2021-01-27:
Thanks the idea, but i hawe the new problem.
ArduLeonardo (ProMicro) CPU: ATMega32U4. Is the circuit, the solder, and not change I2C port. The hardware I2C port use the button and buzzer. But i want to use the A1 and A0 pin. Test is OK. going the I2C communication and beautiful ssd1306 picture. But wire.h is not use! :( please help. 
in which file can i rewrite the port
Thanks Zsolt