MartinL1/BMP280_DEV #11: Problem with data from BMP280

Gennadiy1990, 2020-07-26:
Good day! 
I am using your library for the sensor BMP280. There are problems with the operation of the sensor after switching on. The first two turns on (after writing HEX in 328p Arduino Nano) the sensor works either not correctly or incorrectly at first, but from the third time the work is normal. Please advise to solve this problem. Below is the program code and telemetry.

Thanks!

Code:
#include <BMP280_DEV.h>
#include <Wire.h>
#define PIN_ZUMMER     A1

boolean calibrationBmp = true;
boolean measurementBmp = true;
BMP280_DEV bmp280;

void setup()
{
  float temperature;
  float  pressure;
  float  altitude;
  pinMode(PIN_ZUMMER, OUTPUT);

  Serial.begin(57600);
  Serial.println( "Start");

  if (bmp280.begin(NORMAL_MODE, OVERSAMPLING_X4, OVERSAMPLING_X1, IIR_FILTER_OFF, TIME_STANDBY_05MS) == 0)
  {
    Serial.println( "BMP Error");
    calibrationBmp = false;
    goto Loop;
  }
  delay(100);

  bmp280.getMeasurements(temperature, pressure, altitude);
  Serial.print( "pressure 300-1100 =");  Serial.println( pressure , 2);

  if ( (pressure < 300) || (pressure > 1100))
  { Serial.println( "  pressure ERROR");
    measurementBmp = false;
    goto Loop;
  }

Loop:
  Serial.println( "LOOP");
}

void loop()
{
  float  temperature;            // Create the temperature, pressure and altitude variables
  float  pressure;            // Create the temperature, pressure and altitude variables
  float  altitude;            // Create the temperature, pressure and altitude =variables

  if ((calibrationBmp == false) || (measurementBmp == false))
  {
    if (bmp280.begin(NORMAL_MODE, OVERSAMPLING_X4, OVERSAMPLING_X1, IIR_FILTER_OFF, TIME_STANDBY_05MS) == 0)
    {
      Serial.println( "BMP Error");
      calibrationBmp = false;
      goto Command;
    }
    else
    {
      calibrationBmp = true;
    }
  }

  bmp280.getMeasurements(temperature, pressure, altitude);
  Serial.print( "  pressure 300-1100 =");  Serial.println( pressure , 2);

  if ( (pressure < 300) || (pressure > 1100))
  {
    Serial.println( "  pressure ERROR");
    measurementBmp = false;
    goto Command;
  }

Command:;

  if ((calibrationBmp == false) || (measurementBmp == false))
  {
    digitalWrite(PIN_ZUMMER, HIGH);
    delay(150);
    digitalWrite(PIN_ZUMMER, LOW);
  }

}//END LOOP

Telemetry:
Start
pressure 300-1100 =-0.00
  pressure ERROR
LOOP
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
Start
pressure 300-1100 =-0.00
  pressure ERROR
LOOP
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99

//power off/on
Start
pressure 300-1100 =663.99
LOOP
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.24
  pressure 300-1100 =1004.24
  
  
    //power off/on
  Start
pressure 300-1100 =663.99
LOOP
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =663.99
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.23
  pressure 300-1100 =1004.19
  pressure 300-1100 =1004.19
  pressure 300-1100 =1004.19
  pressure 300-1100 =1004.19
  pressure 300-1100 =1004.19
  

    //power off/on
  Start
pressure 300-1100 =1004.24
LOOP
  pressure 300-1100 =1004.24
  pressure 300-1100 =1004.24
  pressure 300-1100 =1004.24
  pressure 300-1100 =1004.24
  pressure 300-1100 =1004.24
  pressure 300-1100 =1004.24
  pressure 300-1100 =1004.24
  pressure 300-1100 =1004.18
  pressure 300-1100 =1004.18
  pressure 300-1100 =1004.18
  pressure 300-1100 =1004.18
  pressure 300-1100 =1004.18
  pressure 300-1100 =1004.18
  pressure 300-1100 =1004.18
  pressure 300-1100 =1004.15
  pressure 300-1100 =1004.15

Gennadiy1990, 2020-07-26:
I did one more experiment. Removed telemetry. There could be a conflict with Serial. Left only a buzzer to indicate an error. After the first turn on, the same error, on the second turn on it is corrected, on the third the buzzer is silent. If you turn it off for a minute, then щт - everything repeats.

MartinL1, 2020-08-25:
Hi Gennadiy1990,

I ran your code on my Arduino Uno (functionally identical to the Arduino Nano), however the barometer is consistently giving back correct results and I'm unable to reproduce the error that you're experiencing.

Does you breakout board have the correct 5V to 3.3V level shifters between the Arduino Nano and the BMP280 sensor? Otherwise the voltage level thresholds for I2C operation are marginal.

Kind regards,
MartinL1

closed 2020-09-08 by MartinL1

Gennadiy1990, 2020-10-22:
Good day!

I did not notice your answer, for some unknown reason it ended up in spam. I want to say that the problem remains. The levels are matched through a proven circuit on two FETs. There was no such problem with BMP180 when I used it earlier. This cannot be the reason. I got the original sensor from DIGIKEY. The problem has changed somewhat. But every time the power is turned on, several measurements are equal to 0. The number of incorrect results depends on oversampling. There are more for the X4 than for the X1. 
I checked all the settings in your library modes according to the manufacturer's documentation. Everything is correct. Below are two sketches and results. Please review.  I just changed the address of the chip in your library, made the main one with the value of the alternative . One gets the impression that the sensor, after starting for several seconds, is busy with something unclear whether it is being calibrated, or simply stopped. I have read all the documentation for the sensor. Found nothing. There is a library from BOSCH, but it is written in C and is large for 328p.  
I also want to draw your attention to the fact that in the case of zeroing the variables at the beginning of each LOOP (), the results are always alternated with a value equal to 0 (SKETCH1, test 1). 
Thanks!
Arduino IDE v1.8.2

// SKETCH1
#include <BMP280_DEV.h>
#include <Wire.h>
#define PIN_ZUMMER     A1

BMP280_DEV bmp280;

//see test2-floats before setup()
float  temperature = 0;          
float  pressure = 0;          
float  altitude = 0;         

void setup()
{
  pinMode(PIN_ZUMMER, OUTPUT);

  Serial.begin(9600);
  Serial.println( "Start");

  //if (bmp280.begin(NORMAL_MODE, OVERSAMPLING_X4, OVERSAMPLING_X1, IIR_FILTER_OFF, TIME_STANDBY_05MS) == 0)
  if (bmp280.begin(NORMAL_MODE, OVERSAMPLING_X1, OVERSAMPLING_X1, IIR_FILTER_OFF, TIME_STANDBY_05MS) == 0)
  {
    Serial.println( "************* BMP Error ***************");
  }
  delay(100);
  Serial.println( "LOOP");
}

void loop()
{
/* see test1-floats in head of loop() 
float  temperature = 0;          
float  pressure = 0;          
float  altitude = 0;          
*/
  bmp280.getMeasurements(temperature, pressure, altitude);
  Serial.print( "  pressure= ");  Serial.println( pressure , 2);

  if ( (pressure < 300) || (pressure > 1100))
  {
    digitalWrite(PIN_ZUMMER, HIGH);
    delay(100);
    digitalWrite(PIN_ZUMMER, LOW);
  }

}//END LOOP

TESTS:

 test 1
  for define OVERSAMPLING_X1
  float  temperature=0;            
  float  pressure=0;            
  float  altitude=0;            
  in head of loop() result is (NO power off):

  Start
  LOOP
  pressure= 0.00
  pressure= 0.00
  pressure= 0.00
  pressure= 0.00
[... 266 lines cut ...]
Start
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 0.00 temperature= 0.00 altitude= 0.00
 pressure= 1008.98 temperature= 20.47 altitude= 36.12
 pressure= 1008.98 temperature= 20.47 altitude= 36.12
 pressure= 1008.98 temperature= 20.47 altitude= 36.12
 pressure= 1008.98 temperature= 20.47 altitude= 36.12
 pressure= 1008.98 temperature= 20.47 altitude= 36.12
 pressure= 1008.98 temperature= 20.47 altitude= 36.12
LOOP
 pressure= 1008.98 temperature= 20.47 altitude= 36.12
 pressure= 1008.97 temperature= 20.51 altitude= 36.26
 pressure= 1008.97 temperature= 20.51 altitude= 36.26
 pressure= 1008.97 temperature= 20.51 altitude= 36.26
 pressure= 1008.97 temperature= 20.51 altitude= 36.26
 pressure= 1008.97 temperature= 20.51 altitude= 36.26
 pressure= 1008.97 temperature= 20.51 altitude= 36.26
 pressure= 1008.97 temperature= 20.51 altitude= 36.26

Gennadiy1990, 2020-10-22:
I've read other issues. It looks like this issue is due to reading the data ready bit in Normal mode. It is not clear just why to read it. After all, data in this mode is always ready for reading: "3.6.3 Normal mode After setting the mode, measurement and filter options, the last measurement results can be obtained from the data registers without the need of further write accesses." Maybe it makes sense to see how this is implemented in BMP280 API from Bosch Sensortec? 
Thanks!

Gennadiy1990, 2020-10-22:
BST-BMP280-AN000-00_Application_note_self_test.pdf

MartinL1, 2020-10-22:
Hi Gennadiy1990,

The BMP280 barometer has two modes of operation: NORMAL and FORCED mode, (well...actually there's three, if you include SLEEP mode).

NORMAL mode sets up the BMP280 to continously take temperature and pressure measuresments at the standby time interval, while FORCED mode initiates a oneshot single measurement. One thing to bear in mind is that as the BMP280_DEV library is non-blocking, therefore the loop() function needs to continously poll the barometer, in order to detect if the measurement bit in the device has been set (or not). 

Have you tried any of the libraries that I supplied with this library in the examples folder? Do they work correctly with your device?

Gennadiy1990, 2020-10-22:
Hi!
I took your examples as a basis. I sent the tests to you earlier along with my sketches. I can send the library from Bosch. Perhaps they somehow solved this problem with Normal Mode. Your Forsed Mode example works without error.
 Thanks!

MartinL1, 2020-10-22:
@Gennadiy1990 And my example code for the in Normal mode, does that also work?

Looking at your code, there needs to be an _if_ statement enclosing the getMeasurement() function:

```
void loop() 
{
  if (bmp280.getMeasurements(temperature, pressure, altitude))    // Check if the measurement is complete
  {
    Serial.print(temperature);                    // Display the results    
    Serial.print(F("*C   "));
    Serial.print(pressure);    
    Serial.print(F("hPa   "));
    Serial.print(altitude);
    Serial.println(F("m"));  
  }
}
```
This is because the getMeasurements() function is non-blocking and only needs to return data if a new measurement has been received, indicated by changes in the  "measuring" bit. 

In order to detect changes in the "measuring" bit however, the barometer needs to be regularly polled in the loop(). This is due to the fact that on the BMP280, Bosch decided to use a "measuring" bit that's set when the barometer is busy (taking a reading) and cleared when it's not. This is in contrast to their later barometers, such as the BMP388, that use a simple "data ready" bit that's set when the data is ready to be read. It's a subtle but important distinction, the upshot of which is that implementing NORMAL mode on the BMP388 relatively easy, but on the BMP280 its problematic, because the "measuring" bit never actually tells you when the data is available.

Gennadiy1990, 2020-10-23:
Thanks for the quick response.
I haven't checked your Normal Mode yet. Decided to use Forsed Mode. Hopefully it does not block LOOP () in case of any sensor error or data unavailability. 

Then, based on my tests, I can conclude that when the sensor is no power, it seems to "warm up" - it gives out zeros. The documentation states the start time = 2ms. 
And yet it is very interesting how the problem with the normal mode is solved in the Bosch API? Unfortunately, I don't have enough knowledge to find this place in their library.
Thanks!

Gennadiy1990, 2020-10-23:
no power-to switch on

MartinL1, 2020-10-23:
@Gennadiy1990 Yes, Normal mode is non-blocking. The barometer just returns data at the standby time interval.

You mention that the documentation states that the start time is 2ms, however the BMP280_DEV library resets the device and  waits for 10ms before attempting to start initialisation.

Gennadiy1990, 2020-10-23:
@MartinL1   Sorry, I mean the library is reliable in Forsed Mode. The fact is that the program in LOOP () after calling the bmp280.startForcedConversion () function; will skip past the bmp280.getMeasurements function (temperature, pressure, altitude), since the BMP280 with my settings will only prepare the new data after 15ms. 
So on the next pass LOOP () will call the bmp280.startForcedConversion () function again; and the data will not be ready again. BMP 280 will be busy processing new data again. I don't know how you got it to work in Forsed Mode. ) But for me it is very important that working with the sensor does not stop LOOP () for a long time. And also so that under any circumstances the data variables  (T, P, A) do not receive an incorrect value - for example pressure = 0.00 or 600 or  990 instead of 1000. For  LOOP () in my program  it will take about 40-50 milliseconds, together with the BMP280 ( in Forsed Mode). I will repeat once again - the data must always be correct  and correspond to the current atmospheric pressure!!!! 
My device is a parachute ejection system for the UAV.

Gennadiy1990, 2020-10-23:
It may make sense to put the bmp280.startForcedConversion () function; at the beginning of LOOP (), then I will have to read the PWM signal from 1 to 20 milliseconds, then there are some more calculations in LOOP (). And the bmp280.getMeasurements function (temperature, pressure, altitude) near the end of LOOP ()?

MartinL1, 2020-10-24:
@Gennadiy1990 I finally had a chance to sit down with my Arduino Uno and BMP280 breakout board and check what's going on.

The issue is that mechanism required to read the barometer and essential for reading it in Normal mode at the required standby time interval, breaks down when the loop() time exceeds 35ms. You mentioned that your loop() time was 40-50ms.

The reason is that the BMP280, as I mentioned above, uses a "measuring" bit to indicate that the device is busy taking a reading. The problem is that this bit is transient and doesn't go high immediately after initiating a measurement. Also, the BMP280 has no interrupt output, to indicate that data is ready either. This means that it's necessary for the software to poll the measurement bit to detect firstly when there is a low-to-high transition (or rising edge), indicating that barometer is busy and secondly again on a high-to-low transition (or falling edge), indicating that a measurement is complete. If the loop() doesn't poll fast enough the measurement bit transitions are missed causing errors in the measurment reading.

I've also looked at the Bosch library code, but it doesn't appear to be any mechanism to automatically read their own barometer at the standby time interval. They instead have a function that simply calculates the time interval period, so they leave it up to the user to work out the rest. I believe that most other libraries just implement Forced mode.

I'm currently working on a solution that will allow you to read the barometer at the slower loop times. Hopefully, I'll be able to get back to you shortly with solution. You could pepper your loop() code with the bmp280.getMeasurements() function, but that's far from ideal.

MartinL1, 2020-10-24:
@Gennadiy1990 I've added a number of addition functions: getCurrentTemperature(), getCurrentPressure(), getCurrentTempPres(), getCurrentAltitude() and getCurrentMeasurements() to a new release V1.0.17 of the BMP280_DEV library. These allow the barometer to be read with slow loop() times while still maintaining backward compatibillity with the previous version.

These new functions simply request the barometer data without checking the "measuring" bit in the status register. The barometer returns its latest results. This means that the barometer's data registers will be read each time these functions are called, 

Here's an example of my test code for Forced mode:

```
void loop() 
{
  bmp280.startForcedConversion();                 // Start BMP280 forced conversion (if we're in SLEEP_MODE)
  bmp280.getCurrentMeasurements(temperature, pressure, altitude);    // Read the current barometer data
  
  Serial.print(temperature);                      // Display the results    
  Serial.print(F("*C   "));
  Serial.print(pressure);    
  Serial.print(F("hPa   "));
  Serial.print(altitude);
  Serial.println(F("m"));   
  delay(50);                                      // Dummy loop time delay
}
```
and for Normal mode, (simply omit the startForcedConversion() function):
```
void loop() 
{
  bmp280.getCurrentMeasurements(temperature, pressure, altitude);    // Read the current barometer data
  
  Serial.print(temperature);                      // Display the results    
  Serial.print(F("*C   "));
  Serial.print(pressure);    
  Serial.print(F("hPa   "));
  Serial.print(altitude);
  Serial.println(F("m"));   
  delay(50);                                      // Dummy loop time delay
}
```

Gennadiy1990, 2020-10-25:
Thank you for your complete answer! I was working with library version 1.0.15.
  
 But first, I will give two points from the documentation 
(although you probably see them).

"3.9 Data readout
To read out data after a conversion, it is strongly recommended to use a
 burst read and not address every register individually. This will prevent 
 a possible mix-up of bytes belonging to different
 measurements and reduce interface traffic.
 
 3.10 Data register shadowing
In normal mode, measurement timing is not necessarily synchronized to readout. 
This means that new measurement results may become available while the user is 
reading the results from the previous measurement. In this case, shadowing is 
performed in order to guarantee data consistency. Shadowing will only work if 
all data registers are read in a single burst read. Therefore, the user must 
use burst reads if he does not synchronize data readout with the measurement 
cycle. Using several independent read commands may result in inconsistent data.
If a new measurement is finished and the data registers are still being read, 
the new measurement results are transferred into shadow data registers. The content
 of shadow registers is transferred into data registers as soon as the user ends 
 the burst read, even if not all data registers were read. Reading across several
 data registers can therefore only be guaranteed to be consistent within one
 measurement cycle if a single burst read command is used. The end of the burst 
 read is marked by the rising edge of CSB pin in SPI case or by the recognition 
 of a stop condition in I2C case.
 After the end of the burst read, all user data registers are updated at once."
 
I have reviewed your changes in the library. When using new functions, the ability
 to get information about the correct execution of functions was lost
 void BMP280_DEV :: getCurrent ....... (no return). So, based on my requirements 
 for the correctness of the data, I have a couple of questions. 
 
1. Are you sure that the Wire.h library (which you are using) functions for reading the temp_ and press_ data registers strictly provide a "single burst read" mode as 
described in section "5.2.2 I²C read" of the BMP280 documentation? 
This is very important for the guarantee data consistency! Unfortunately, I do not
 know this library very well and it is difficult for me to understand this issue.
 2. I came to the conclusion that in order to use the new functions getCurrent ....... 
it is necessary to use only NORMAL mode as it has a "shadowing" mode. Hopefully it 
will provide data consistency!
3. I have a begin () function in LOOP () for restarting the sensor if the previously 
obtained pressure value does not satisfy the range of 300 ... 1100 hPa. Therefore, 
I cannot use an IIR filter (X2 or more). It will give a 99.9% accurate result only 
after several (4-7 or more) measurements after 
a restart. Do you think I am right? 

I will say that after all these works, I stopped trusting Bosch! :-(((
Thanks!

MartinL1, 2020-10-25:
I'll investigate each point in turn and get back to you later on today.

MartinL1, 2020-10-25:
@Gennadiy1990 

1. Looking at my oscilloscope with the I2C decoder, I can see that the Wire library does a single burst read:

[image]

As you can see my Arduino Uno performs a read request to the BMP280 at address 0x77, which responds with 6 continous bytes of temperature and pressure data.

2. Yes, reading the datasheet it's clear that shadow registers allow the data to be read in normal mode, provided the data is read using a single burst read, as in the above image. What's not so clear, is whether it's possible to use the shadow registers in forced mode, although reading between the lines it most probably is possible, in tests I don't get any data corruption when reading the barometer as fast as possible in forced mode, just repeated results. In any case, if your loop() time is greater than your barometer conversion time, the data is guaranteed to be ready anyway. 

I'm not sure how the Bosch BMP280's designers expect users determine that a forced mode conversion is ready, especially if you're unable to poll the "measuring" bit at a sufficient rate? 

3. I've never tried calling begin() from the loop(), but it should work, as the BMP280_DEV library will reset the device in an event. Also, after reset the first reading I'm getting is junk. The measurements thereafter are good with the device temperature gradually rising due continuous operation.

Around 6 to 7 years ago, Bosch MEMS barometers, such as the BMP085, were to be honest not that good. The measurements weren't very precise and required a lot of filtering in software. Nowadays, they've improved immeasurably and their latest offerings, such as the BMP388 for example, are excellent. These devices are housed in a tiny package, have good precision, internal filtering and support an unlatched or latched "data ready" status bit, plus an external interrupt pin. The BMP388 even has a 512 byte internal memory for data logging over extended periods. The BMP280 is also very good despite its shortcomings.

Gennadiy1990, 2020-10-26:
Good afternoon!
Thank you so much for the experiment. I didn’t expect you to pay so much attention to this issue. My oscilloscope cannot decode I2C. 
I will rewrite my sketch, check it at work. I will report on the results. At the same time, I will accurately measure the cycle time in my program.
 I also tested BMP085 some time ago (in 2014). Then BMP180. 

Thanks!

Gennadiy1990, 2020-10-28:
Good afternoon!
Thank you for your help! It is very important for me!
 I tested the sensor in two sketches, restarting through begin () and switching power. Inserted a millisecond counter in the first column. Below are the test results and a sketch.
 The initial appearance of data occurs approximately 12 milliseconds after the startNormalConversion () function is executed. IIR_FILTER_OFF does not affect data flow. Oddly enough, new data appears every 5 milliseconds. It seems that the BMP280 does all the processing in less than 1 millisecond, and then rests for 5 milliseconds. I will write my working sketch in an abridged version, taking into account the correction of the delay in issuing the correct data, and if you want, I will send it to you. It is very important for me to know your opinion! 

#include <BMP280_DEV.h>                           // Include the BMP280_DEV.h library
#define PIN_ZUMMER     A1
#define P_MAX          1100.0
#define P_MIN           300.0

BMP280_DEV bmp280;                                // Instantiate (create) a BMP280_DEV object and set-up for I2C operation (address 0x77)
float temperature, pressure, altitude;            // Create the temperature, pressure and altitude variables
unsigned long currentMillis = 0;//
void setup()
{
  pinMode(PIN_ZUMMER, OUTPUT);
  Serial.begin(57600);                           // Initialise the serial port
}

void loop()
{
  if (bmp280.begin() == 0)                                // Default initialisation, place the BMP280 into SLEEP_MODE
  {
    Serial.println( "************* BMP Error ***************");
  }
  bmp280.setPresOversampling(OVERSAMPLING_X4);    // Set the pressure oversampling to X4
  bmp280.setTempOversampling(OVERSAMPLING_X1);    // Set the temperature oversampling to X1
  bmp280.setIIRFilter(IIR_FILTER_2);              // Set the IIR filter to setting 4
  bmp280.startNormalConversion() ;  // Start continuous measurement in NORMAL_MODE

  for (int i = 0; i < 12; i++)
  {
    bmp280.getCurrentMeasurements(temperature, pressure, altitude);    // Check if the measurement is complete
    currentMillis =  millis();
    Serial.print (currentMillis);
    Serial.print ("\t");
    Serial.print(temperature, 3);                   // Display the results
    Serial.print("\t ");
    Serial.print(pressure, 3);
    Serial.print("\t ");
    Serial.println(altitude, 3);
    if ((pressure < P_MIN) || (pressure > P_MAX ))
    {
      digitalWrite(PIN_ZUMMER, HIGH);
      delay(100);
      digitalWrite(PIN_ZUMMER, LOW);
    }
  }

}

// bmp280.begin() in Loop()
//tests 
//power on
12	26.260	 675.744	 3689.781
15	26.260	 675.744	 3689.781
18	26.260	 675.744	 3689.781
21	26.260	 675.744	 3689.781
24	22.930	 1009.054	 35.806
28	22.930	 1009.054	 35.806
32	22.930	 1009.054	 35.806
37	22.940	 1009.050	 35.840
43	22.940	 1009.050	 35.840
48	22.950	 1009.060	 35.749
53	22.950	 1009.060	 35.749
57	22.950	 1009.060	 35.749
74	26.260	 675.744	 3689.781
77	26.260	 675.744	 3689.781
80	26.260	 675.744	 3689.781
83	26.260	 675.744	 3689.781
[... 87 lines cut ...]
//power off/on
12	26.260	 675.744	 3689.781
15	26.260	 675.744	 3689.781
18	26.260	 675.744	 3689.781
21	26.260	 675.744	 3689.781
24	23.410	 1008.429	 41.238
28	23.410	 1008.429	 41.238
32	23.410	 1008.429	 41.238
37	23.420	 1008.426	 41.266
43	23.420	 1008.426	 41.266
48	23.430	 1008.409	 41.414
53	23.430	 1008.409	 41.414
57	23.430	 1008.409	 41.414
62	23.450	 1008.440	 41.151
67	23.450	 1008.440	 41.151
72	23.460	 1008.423	 41.304
77	23.460	 1008.423	 41.304
81	23.460	 1008.423	 41.304
87	23.470	 1008.412	 41.398
92	23.470	 1008.412	 41.398
97	23.490	 1008.417	 41.352
102	23.490	 1008.417	 41.352
107	23.490	 1008.417	 41.352
112	23.500	 1008.400	 41.506
117	23.500	 1008.400	 41.506

MartinL1, 2020-10-31:
@Gennadiy1990 Your results are consistent with what I'd expect to see. 

The BMP280_DEV begin() functon resets the barometer and generates a 10ms delay to allow the device to settle. This is  followed by device initalisation and transmitting the results to the console over the serial port, which most likely accounts for the remain 2ms of the 12ms that you are measuring. 

The initial measurement that the barometer generates after reset is junk. The barometer in normal mode takes a new measurement every 5ms. The time taken to produce a new measurement after reset is 10ms + 2ms + 5ms = 17ms. These results are again consistent with what is being displayed on your console.