MartinL1/BMP280_DEV #13: BMP280_I2C_Normal error when compiling on ESP32.

chbll, 2020-08-30:
Hi all, 
getting the following errors when attempting to compile BMP280_Dev on an ESP32 (TTGO_SIM800L). Exit Status 1. 
Latest IDE + BMP280_Dev versions being used.

Code works fine on a Nano with a single BMP280. Using bmp280.begin(0x76); as I2C scanner has picked up breakout bd on this address, using A4/A5 5V. Temp and Pressure seem accurate.

Using 21/22 5v on TTGO.
Any ideas?
cheers
chris

"C:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protobuf-c" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protocomm" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/pthread" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/sdmmc" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/soc" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spi_flash" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spiffs" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/tcp_transport" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/tcpip_adapter" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ulp" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/vfs" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wear_levelling" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wifi_provisioning" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/wpa_supplicant" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/xtensa-debug-module" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-face" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32-camera" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-face" "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10813 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\\Users\\chris.blundell\\AppData\\Local\\Arduino15\\packages\\esp32\\hardw [... cut ...]
   data = spi->transfer(data);
                       ^
cc1plus.exe: some warnings being treated as errors
Using library BMP280_DEV at version 1.0.15 in folder: C:\Users\chris.blundell\Documents\Arduino\libraries\BMP280_DEV 
Using library Wire at version 1.0.1 in folder: C:\Users\chris.blundell\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire 
Using library SPI at version 1.0 in folder: C:\Users\chris.blundell\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI 
exit status 1

lockie555, 2020-08-31:
I have a similar problem.
When will you make a modification to allow user-defined pins for I2C operation on the ESP32.

MartinL1, 2020-09-06:
Hi Chris,

I've tried compiling for the ESP32 and compilation succeeded without any errors or warnings. I selected "ESP32 Dev Module" as the board in the Arduino IDE.

The error report appears to be complaining about the SPI constructor. Are you using the example sketch for the I2C port?

MartinL1, 2020-09-08:
@lockie555 I've now modified the BMP280_DEV library to allow user-defined pins for I2C operation on the ESP32.

I'm now closing this issue.

closed 2020-09-08 by MartinL1

ab-it, 2020-11-06:
The complaints while compiling are about the order in which things are initialised. Initialise them in the order they are defined.  I was able to remove all errors in a local copy. However, while copy/paste'ing things I seem to have damaged the comments (not just tab/space, but also some weird quotes, maybe UTF?) so I do not just want to dump the files here.

When I see these lines:
C:\Users\chris.blundell\Documents\Arduino\libraries\BMP280_DEV/Device.h:77:11: error: 'Device::spiPort' will be initialized after [-Werror=reorder]
uint8_t spiPort; // SPI port type VSPI or HSPI
^
C:\Users\chris.blundell\Documents\Arduino\libraries\BMP280_DEV/Device.h:73:13: error: 'SPIClass* Device::spi' [-Werror=reorder]
SPIClass* spi; // Pointer to the SPI class

I knew to move "SPIClass* spi;" above "uint8_t spiPort; ".

There are about 3 of these.  I also got the warning (which is turned into an error in my setup) about data possibly not being defined before use, easy to fix:   s/uint8_t data;/uint8_t data=0;/

MartinL1, 2020-11-07:
@ab-it I've tried compiling examples using the Arduino IDE on ESP32 target boards for both VSPI and HSPI ports, but I'm not getting any errors.

Might I ask which ESP32 target board you're using and which SPI port you've configured (VSPI or HSPI)?

ab-it, 2020-11-07:
https://stackoverflow.com/questions/12222417/why-should-i-initialize-member-variables-in-the-order-theyre-declared-in

I'm using doit esp32 devkit v1. 
I don't use the SPI port, but it is compiled anyways.

Not 100% sure this matters but it seem to: in my preferences file I see compiler.warning_level=all

MartinL1, 2020-11-08:
@ab-it Thanks for perservering with the issue and for the additional link explaining the situation, I found it really helpful.

After switching the compiler warning level from "default" to "all" in preferences as you suggested, then re-starting the Arduino IDE, I managed to get the same error messages.

I've now reordered the device class member variables to match the initialisation in the constructor and generated a new releases of both my BMP280_DEV (Version 1.0.18) and BMP388_DEV (Version 1.0.3) libraries.

Thank you for your support. I'll have to remember to leave my Arduino IDE compiler settings set to "all" when developing code in the future.