MartinL1/BMP280_DEV #15: How to change I2C pins Leonardo (32U4)

kreamuhely, 2021-01-27:
I have the new problem wich 32U4 cpu.
I have ArduLeonardo (ProMicro) CPU: ATMega32U4. Is the circuit soldered, and i didn't change I2C port. The hardware I2C port to use button (pin 2) and buzzer (pin 3). But i want to use I2C the port A1 and A0 pin . Test is OK. going the I2C Port A1 and 0, communication good and beautiful ssd1306 picture. But wire.h is not use! :( please help.
in which file can i rewrite the port
Thanks Zsolt

MartinL1, 2021-01-28:
Hi Zsolt,

Unfortunately, the custom I2C pins feature is only available on the ESP8266, as it doesn't have an on-chip, hardware I2C module. The Atmega32U4 based boards have fixed pins for I2C, namely SCL and SDA. This is because the BMP280_DEV code relies on the Arduino Wire library for I2C communications.

rpehlm, 2021-03-21:
Using an ESP32 Devkit C v4 (38pin board), supplied by AZ-Delivery.de with an Adafruit BMP280
Trying to use alternate SDA and SCL pins.
The example given is: BMP280_DEV bmp280(A6, A7);	// Set up I2C communications on ESP32 pins A6 (SDA) and A7 (SCL): bmp280(SDA, SCL);
I've found that A6 = 34 and A7 = 35 - via serial print :)
Tried connecting the BMP280 SDI pin to the ESP32 GPIO pin 34 and SCK to pin 35.
Didn't work.
I don't think I'm understanding the A6 and A7 pin designations.
Tried searching for ESP32 A6 and A7 pin information and found nothing.
Example I'm using is here (renamed as .txt so it can be dropped here):
[bmp280test.txt](https://github.com/MartinL1/BMP280_DEV/files/6177498/bmp280test.txt)
Any ideas where I'm going wrong?

rpehlm, 2021-03-21:
Darn, dropped the above comment in the wrong place!
Will start a new thread if required - my bad!

MartinL1, 2021-03-22:
@rpehlm The reason why ESP32 pins GPIO34 and GPIO35 aren't working, is because GPIO34, 35, 36 & 39 should only be used as inputs: [https://randomnerdtutorials.com/esp32-pinout-reference-gpios/](url). 

The A6 and A7 designations are the ESP32 Arduino pins, however it's also possible to just specify the GPIO pin numbers as arguments for the bmp280(SDA, SCL) constructor (function) as well. On my Adafruit ESP32 Feather board, A6 is on GPIO14 and A7 on GPIO32, which differs from your ESP32 Devkit C v4 board. 

The Arduino pin specifications for the various ESP32 boards can be found in their "pins_arduino.h" files, which on my machine are located here:

C:\Users\Computer\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\variants\...

rpehlm, 2021-03-22:
Hi Martin,
Thanks for that information.  Got the issue fixed :)
Now using GPIO pins 25 and 26.

MartinL1, 2021-03-22:
Hi rphelm,
Glad to here that it's now working.

closed 2021-03-22 by MartinL1